rust-osdev/uefi-rs #1458: Add function that removes a printed character from the screen.

Motanescu1357, 2024-11-03:


nicholasbishop, 2024-11-03:
You can emit a backspace character (numeric value of 8). Here's a simple example, which I tested in both a VM on a Lenovo Thinkpad X1 Carbon:

```rust
#![no_main]
#![no_std]

use uefi::prelude::*;
use uefi::print;

#[entry]
fn main() -> Status {
    uefi::helpers::init().unwrap();

    print!("abcdefghijklmnopqrstuvwxyz");

    for _ in 0..26 {
        // Pause briefly.
        boot::stall(500_000);

        // Print backspace character.
        print!("\x08");
    }

    Status::SUCCESS
}
```

Motanescu1357, 2024-11-04:
for me it does not work.

Motanescu1357, 2024-11-04:
Maybe my bios does not implement special characters.

nicholasbishop, 2024-11-04:
Checking a couple things:
1. What specific machine are you testing on?
2. Did you test exactly the code above, or some variation? (I want to make sure we're both testing the exact same thing.)

Motanescu1357, 2024-11-04:
I used a asus laptop.
I tested the same code.

nicholasbishop, 2025-02-09:
Unfortunately I don't think there's anything we can do here. It sounds like the vendor's implementation just doesn't support this. As an alternative you could potentially use the [`GraphicsOutput`](https://docs.rs/uefi/latest/uefi/proto/console/gop/struct.GraphicsOutput.html) protocol to fully replace the console output with manually-drawn text.

closed 2025-02-09 by nicholasbishop