rust-osdev/uefi-rs #1470: Relicense uefi-rs under dual MIT/Apache-2.0

nicholasbishop, 2024-11-10:
To improve compatibility with the Rust ecosystem, we are planning to relicense all the code in the uefi-rs repo under the MIT/Apache-2.0 licenses.

This will affect the `uefi-raw`, `uefi-macros`, and `uefi` crates, as well as all other code in this repo.

We need all contributors to agree to this license. (If any contributors do not agree, we may still proceed with the relicense, but with those contributions removed.) Note that this relicensing will not affect existing releases; they will always be available under the existing MPL-2.0 license.

To agree to this relicense, please leave a comment with the following message:

```
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.
```

Thanks all for your help with this!

## Contributors

- [ ] @3than3
- [ ] @AdrianDanis
- [ ] @Andy-Python-Programmer
- [ ] @AtsukiTak
- [ ] @BinaryTENSHi
- [ ] @Bobo1239
- [ ] @ColinFinck
- [ ] @Freax13
- [ ] @FredrikAleksander
- [ ] @GabrielMajeri
- [ ] @HTGAzureX1212
- [ ] @HadrienG2
- [ ] @IsaacWoods
- [ ] @JarlEvanson
- [ ] @JeffLi01
- [ ] @JohnAZoidberg
- [ ] @JonahPlusPlus
- [ ] @LightAndLight
- [ ] @MaulingMonkey
- [ ] @NathanRoyer
- [ ] @RaitoBezarius
- [ ] @Redrield
- [ ] @Stzx
- [ ] @YtvwlD
- [ ] @andre-braga
- [ ] @baloo
- [ ] @bjadamson
- [ ] @bjorn3
- [ ] @blitz
- [ ] @cmoylan
- [ ] @ctrlaltmilk
- [ ] @d-sonuga
- [ ] @dependabot-preview[bot]
- [ ] @dependabot[bot]
- [ ] @devsnek
- [ ] @diekmann
- [ ] @e820
- [ ] @edigaryev
- [ ] @felipebalbi
- [ ] @foxcob
- [ ] @gelven4sec
- [ ] @gil0mendes
- [ ] @gurry
- [ ] @gz
- [ ] @hughsie
- [ ] @iankronquist
- [ ] @imtsuki
- [ ] @josephlr
- [ ] @julic20s
- [ ] @kendase3
- [ ] @liferooter
- [ ] @medhefgo
- [ ] @necauqua
- [ ] @nicholasbishop
- [ ] @nkaretnikov
- [ ] @nsemmel
- [ ] @ocadaruma
- [ ] @pavlus
- [ ] @phil-opp
- [ ] @phip1611
- [ ] @pratyai
- [ ] @raccog
- [ ] @renovate[bot]
- [ ] @rexlunae
- [ ] @roblabla
- [ ] @sagudev
- [ ] @sdemos
- [ ] @sky5454
- [ ] @sreehax
- [ ] @stevefan1999-personal
- [ ] @stillinbeta
- [ ] @supdrewin
- [ ] @timrobertsdev
- [ ] @toku-sa-n
- [ ] @tomoyat1
- [ ] @upsuper
- [ ] @veluca93
- [ ] @wangrunji0408
- [ ] @zdivelbiss
- [ ] @zhangpf

nicholasbishop, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

Freax13, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

JarlEvanson, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

gz, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

baloo, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

Andy-Python-Programmer, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

RaitoBezarius, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0
license, allowing licensees to choose either at their option.

HTGAzureX1212, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

ctrlaltmilk, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

bjadamson, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

nicholasbishop, 2024-11-10:
(I think Github only pinged the first 50 names in the description, so pinging the rest here.)

To improve compatibility with the Rust ecosystem, we are planning to relicense all the code in the uefi-rs repo under the MIT/Apache-2.0 licenses.

To agree to this relicense, please leave a comment with the following message:

```
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.
```

- @liferooter
- @medhefgo
- @necauqua
- @nicholasbishop
- @nkaretnikov
- @nsemmel
- @ocadaruma
- @pavlus
- @phil-opp
- @phip1611
- @pratyai
- @raccog
- @renovate[bot]
- @rexlunae
- @roblabla
- @sagudev
- @sdemos
- @sky5454
- @sreehax
- @stevefan1999-personal
- @stillinbeta
- @supdrewin
- @timrobertsdev
- @toku-sa-n
- @tomoyat1
- @upsuper
- @veluca93
- @wangrunji0408
- @zdivelbiss
- @zhangpf

roblabla, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

toku-sa-n, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

veluca93, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

upsuper, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

LightAndLight, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

stevefan1999-personal, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

pratyai, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

ocadaruma, 2024-11-10:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

iankronquist, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0
license, allowing licensees to choose either at their option

Ian Kronquist

zdivelbiss, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

gelven4sec, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

raccog, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

felipebalbi, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

timrobertsdev, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

e820, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

JonahPlusPlus, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

StevenDoesStuffs, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

sreehax, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

JohnAZoidberg, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

gil0mendes, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

liferooter, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to chose either at their option.

julic20s, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

MaulingMonkey, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

sagudev, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

phip1611, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

GabrielMajeri, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

necauqua, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

edigaryev, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

ColinFinck, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

devsnek, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

bjorn3, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

BinaryTENSHi, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

phil-opp, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

hughsie, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

gurry, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

HadrienG2, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

d-sonuga, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

nkaretnikov, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

diekmann, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

Stzx, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

blitz, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

IsaacWoods, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

wangrunji0408, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

NathanRoyer, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

andre-braga, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

josephlr, 2024-11-11:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

AtsukiTak, 2024-11-12:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

foxcob, 2024-11-12:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

kendase3, 2024-11-12:
I license past and future contributions under the dual MIT/Apache-2.0 license, allowing licensees to choose either at their option.

[19 more replies]

closed 2025-02-07 by nicholasbishop