rust-osdev/uefi-rs #1572: Unable to use GOP

SniverDaBest, 2025-03-13:
I'm new to using this crate, and I'm trying to get some simple graphics working. However, whenever I try and get the `GOP`, it locks up the system, without a panic, and I couldn't figure out why. Hopefully this is the right place for this...

Here's *most* of the code I have
```
pub const LEMONCAKE_VER: &str = "25m3-UEFI";
use log::{error, info};
use uefi::{
    helpers,
    prelude::*,
    proto::console::gop::GraphicsOutput,
};

#[entry]
fn main() -> Status {
    helpers::init().unwrap();
    info!("Running Lemoncake {}", LEMONCAKE_VER);

    let gop_handle =
        boot::get_handle_for_protocol::<GraphicsOutput>().expect("Unable to find the GOP!");
    info!("This does appear!");
    let mut gop = boot::open_protocol_exclusive::<GraphicsOutput>(gop_handle);
    info!("This does NOT appear!");

    return Status::SUCCESS;
}

#[panic_handler]
fn panic(info: &core::panic::PanicInfo) -> ! {
    error!(
        "(X_X)\n\nUh-Oh, Lemoncake panicked!\nMessage: {}\nLocation: {}@L{}:{}",
        info.message(),
        info.location().unwrap().file(),
        info.location().unwrap().line(),
        info.location().unwrap().column()
    );

    loop {}
}
```

And as one may be able to tell, the "This does NOT appear!" text does not appear.

[image]

I've tried quite a few things, and nothing works. Even the sierpinski example inside of here doesn't work when putting it into my project.

If it helps, I'm running openSUSE Tumbleweed, and how I've been running QEMU.
```
qemu-system-x86_64 -enable-kvm \
    -bios bin/ovmf.bin\
    -drive format=raw,file=fat:rw:bin/esp\
    -m 256M
```

I don't know if this is just my bad programming, or if something's possibly wrong with the crate itself *(which i doubt)*. However, I would like to get this resolved soon.

nicholasbishop, 2025-03-13:
I think that what's happening is that since you are opening the gop in exclusive mode, the logger can no longer write to the screen. If you send logs to the serial console instead, and configure qemu to send serial to stdout, I suspect the missing log will show up.

SniverDaBest, 2025-03-13:
Wow, thanks for answering so quickly! I just checked the serial console, and since everything printed through info! and error! apparently goes to serial automatically, I can see all of the logs. Thanks so much for the help!