rust-osdev/uefi-rs #1575: RFC: use `core::net` IP address types in `uefi`'s public API

nicholasbishop, 2025-03-16:
Currently, we provide the following IP address types:

* `uefi-raw`:
  * [`struct Ipv4Address`](https://docs.rs/uefi-raw/latest/uefi_raw/struct.Ipv4Address.html) - 4 raw bytes
  * [`struct Ipv6Address`](https://docs.rs/uefi-raw/latest/uefi_raw/struct.Ipv6Address.html) - 16 raw bytes
  * [`union IpAddress`](https://docs.rs/uefi-raw/latest/uefi_raw/union.IpAddress.html) - untagged union, can be either v4 or v6 address, but no way to tell without out-of-band data
* `uefi`:
  * [`struct IpAddress`](https://docs.rs/uefi/latest/uefi/proto/network/struct.IpAddress.html) - 16 raw bytes. Compatible with the `uefi_raw::IpAddress` union, but implemented differently.  No way to tell which type of address it is without out-of-band data

* And these builtin types are in `core::net`:
  * [`struct Ipv4Addr`](https://doc.rust-lang.org/core/net/struct.Ipv4Addr.html) - opaque struct, but essentially 4 raw bytes
  * [`struct Ipv6Addr`](https://doc.rust-lang.org/core/net/struct.Ipv6Addr.html) - opaque struct, but essentially 16 raw bytes
  * [`enum IpAddr`](https://doc.rust-lang.org/core/net/enum.IpAddr.html) - tagged enum containing either `Ipv4Addr` or `Ipv6Addr`

The types in `uefi-raw` correctly match the spec, so there's nothing we need to change there.

In `uefi` however, I think it might make sense to switch to the `core::net` types in the public API, and drop our `IpAddress`. The `core::net` types are not ABI-compatible with the UEFI types, so internally we'll need to translate between them. This should be a very low-cost operation though.

Using the `IpAddr` enum provides a better experience than using an untagged enum that requires a separate flag to indicate which type of address it contains. We could define our own enum of course, but using `core::net` makes it easier to integrate with other Rust code by using a shared type.

This change would be an API break, but it only affects a fairly small part of the library (the `SimpleNetwork` and `pxe::BaseCode` protocols).

phip1611, 2025-03-16:
Yep, sounds good! Thanks for the write-up

phip1611, 2025-04-21:
At first, I wanted to outline my ideas here. Then I noticed it is simpler to put it into code to see if it works out. My attempt is here: https://github.com/rust-osdev/uefi-rs/pull/1645

closed 2025-10-21 by phip1611