rust-osdev/uefi-rs #1604: exit_boot_services() rename

ofluffydev, 2025-04-06:
The `exit_boot_services()` function requires a MemoryType where it should not, and returns a memory map. I believe this function does more if not something entirely different from what you would expect exiting UEFI mode to be, therefore it needs to be renamed, or fixed to not require a MemoryType. I believe it used to be that way and was changed for some reason. If you're seriously going to keep it this way, at least help my writing better docs on both the `enums` and `exit_boot_services()` so people can understand what it means.

Yes "document it youself" blah blah I so WOULD PR better docs right now if I could understand this, it would be way better for the creators/maintainers to document them rather than wait months for me to figure it out just now learning UEFI and OS development with Rust.

nicholasbishop, 2025-04-06:
I can tell you're frustrated! I saw you just filed a similarly frustrated issue in the bootloader crate as well. Please take a little time to calm down before returning to this, otherwise we can't have a useful conversation. Thanks!

ofluffydev, 2025-04-06:
did you read what I said?.. You don't name something `color_this_red` and it color said thing purple. The name doesn't make since or at least needs an explanation. It can be better explained in the comments or renamed altogether to fix this...

nicholasbishop, 2025-04-06:
Please take some time to calm down. if you continue to behave rudely, we'll ban you.

ofluffydev, 2025-04-06:
Sincerely sorry if anything is coming off as rude, I am only asking to rename the function, explain it better in docs, or explain in this issue why it is in it's state for future reference.

phip1611, 2025-04-06:
Hey @ofluffydev,

I've set up #1605 to improve the situation. 

In short: Exiting UEFI boot services requires a **non-trivial sequence of steps**,
including safe retrieval and finalization of the memory map. To clarify
why our exit_boot_services function is more than a thin wrapper, the
documentation has been updated accordingly.

closed 2025-04-07 by nicholasbishop