rust-osdev/uefi-rs #1608: uefi: release 0.35

phip1611, 2025-04-06:
It's been over two months since our last release and we already have lots of great additions, changes, and fixes in the pipeline.

I've added a few things that I'd definitely like to see in the new release:

## Release blockers
- [ ] #1606
- [ ] #1599
- [ ] #1605

## Maybe release blockers
- [ ] #1603
- [ ] #1589 
- [ ] #1595

nicholasbishop, 2025-05-02:
I'm now back from a short hiatus -- I got married last weekend! 😁 

What do you think about just going ahead and cutting a release for uefi-0.35 and uefi-raw-0.11? I know we still have a number of unmerged PRs, some of which need my review, but as you noted it's been a while since we last released and there are a lot of changes there already. I intend to get to some of that review work soon, but I don't want to hold up this release any longer.

phip1611, 2025-05-03:
Fantastic news! And if you’re taking a bit more time off instead of jumping back into uefi-rs—totally understandable 😂