rust-osdev/uefi-rs #1616: cargo xtask run fails to fetch edk2

kraxel, 2025-04-10:
```
[INFO] downloading https://github.com/rust-osdev/ovmf-prebuilt/releases/download/edk2-stable202402-r1/edk2-stable202402-r1-bin.tar.xz
Error: remote request failed

Caused by:
    io: Peer disconnected
```
Trying to fetch the file with wget shows that github redirects to `https://objects.githubusercontent.com/...`.  I think this is new, I can't remember having seen this before.

kraxel, 2025-04-10:
Even when using `--ovmf-code` + `--ovmf-vars` xtask tries to fetch prebuild edk2.

phip1611, 2025-04-10:
Good catch! Thanks.

1.) Probably we have to update the URL
2.) it shouldn't fetch when we provide OVMF as environmental variable

nicholasbishop, 2025-05-07:
I think the download issue must have been a transient problem with github or the network; it seems to be working fine now.

I also checked on why it's downloading if you specify `--ovmf-code + --ovmf-vars`: it's because we need a third file, `shell.efi`. If you specify that one as well (with `OVMF_SHELL`, since there's no arg defined for it), the download is correctly skipped.

So I think the only thing we should maybe do here is add a `--ovmf-shell` arg, to make that 3rd dependency more discoverable in the help output.