rust-osdev/uefi-rs #1617: snp test is broken

kraxel, 2025-04-10:
the snp test depends on the pxe test having configured the network beforehand, it does not work on its own.
swap the order of the tests ...
```
 pub fn test() {
     info!("Testing Network protocols");
 
-    pxe::test();
     snp::test();
+    pxe::test();
 }
```
... and watch it fail:
```
[PANIC]: panicked at uefi-test-runner/src/proto/network/snp.rs:112:9:
assertion `left == right` failed
  left: [0, 0, 0, 0, 0]
 right: [4, 4, 3, 2, 1]
Error: tests did not complete successfully
```

phip1611, 2025-04-10:
Oh, good catch! It shouldn't be like that; test runs should be idempotent

phip1611, 2025-04-15:
Maan, I was spending way too much time on this. Here is what is happening:

- PXE test assigns IPv4 to network interface
- SNP test can send IPv4 packet

Otherwise, the SNP test receives an ARP packet instead of the IPv4 packet.

We have two options:
- use DHCP in SNP test (meh, dislike)
- send raw ethernet frame without IPv4 to ourselves (currently working on it, but not as simple as expected)

phip1611, 2025-04-15:
I'm stopping the work on this. I've spent many hours into it, but I didn't get either way running reliably.

Didn't think that it is not possible to send a raw Ethernet frame to ourself..