rust-osdev/uefi-rs #1710: Build Error: Cannot resolve `uefi::prelude::*` items (SystemTable, println!, etc.) and `#[entry]` macro issues with v0.35.0 on nightly"

deepm1nd, 2025-06-23:
**Describe the bug**
When attempting to build a minimal UEFI application using `uefi-rs` v0.35.0 on a recent nightly Rust toolchain (specifically targeting `x86_64-unknown-uefi` with `-Zbuild-std` for `core` and `compiler_builtins`), basic items from `uefi::prelude::*` (such as `println!`, `SystemTable`, `Boot`, `image_handle()`, `system_table()`) are not found by the compiler. Additionally, the `#[entry]` macro seems to have issues:
- If `efi_main` is defined to take `Handle` and `SystemTable<Boot>` arguments (as per documentation), the build fails with "Entry function must have no arguments".
- If `efi_main` is defined to take no arguments (and attempts to use the free functions `image_handle()` and `system_table()`), it fails because those functions (and other prelude items) are not found.

This issue is reproducible in a minimal example and also blocks a more complex project.

**Environment:**
*   `uefi-rs` version: `0.35.0`
*   Rust toolchain: `nightly` (specifically `1.89.0-nightly (be19eda0d 2025-06-22)` as per build log, though the `rust-toolchain.toml` just specifies `nightly`)
*   Target: `x86_64-unknown-uefi`
*   Build host OS (if relevant, this is a sandboxed environment): Linux-based
*   `build-std` configuration: `build-std = ["core", "compiler_builtins"]` with `build-std-features = ["compiler-builtins-mem"]`

**To Reproduce**
Steps to reproduce the behavior:

1.  Set up the following minimal project structure and files:

    *   `uefi_minimal_test/Cargo.toml` 
    *   `uefi_minimal_test/.cargo/config.toml` 
    *   `uefi_minimal_test/rust-toolchain.toml` 
    *   `uefi_minimal_test/src/main.rs`

2.  Ensure the target and `rust-src` are installed:
    ```bash
    rustup target add x86_64-unknown-uefi
    rustup component add rust-src --toolchain nightly 
    ```
    (These steps were performed in the environment)

3.  Run the build command from the directory containing `uefi_minimal_test`:
    ```bash
    cargo +nightly build --manifest-path uefi_minimal_test/Cargo.toml --target x86_64-unknown-uefi 
    ```
    (Note: `-Zbuild-std` is implicitly handled by `.cargo/config.toml`)

**Expected behavior**
The minimal application should compile successfully. If `efi_main` takes arguments, `#[entry]` should handle it. If `efi_main` takes no arguments, items from `uefi::prelude::*` like `println!`, `image_handle()`, `system_table()`, `SystemTable`, `Boot` should be resolved.

**Actual behavior**
The build fails with errors indicating that items from `uefi::prelude::*` cannot be found. For example:
- `error: cannot find macro \`println!\` in this scope`
- `error[E0425]: cannot find function \`image_handle\` in this scope`
- `error[E0425]: cannot find function \`system_table\` in this scope`
- `error[E0433]: failed to resolve: use of undeclared type \`SystemTable\``
- `error[E0412]: cannot find type \`Boot\` in this scope`

(Full build log provided in `uefi_minimal_test_build_log.txt` attachment)

If `src/main.rs` is changed so `efi_main` takes `_image_handle: Handle, mut system_table: SystemTable<Boot>` arguments, the error changes to:
`error: Entry function must have no arguments`

**Additional context**
*   This issue was encountered while trying to set up a more complex kernel project (`torux`), and the minimal example was created to isolate the problem. The `torux` project experiences the same `uefi` module/item resolution failures.
*   The `torux` project also consistently fails with "no global memory allocator found but one is required", even with a `#[global_allocator]` defined and `extern crate alloc;`. This error also appears in the minimal example if `alloc` types are used and `uefi = { features = [\"alloc\"]}` is enabled alongside `build-std = [..., \"alloc\"]`. For the provided minimal example, `alloc` usage was removed to focus on the prelude/entry macro issue.
*   The problem persists across various attempts to configure `build-std` (with/without `"alloc"`) and `uefi` crate features.
*   The `cargo tree` for the minimal example (and the more complex project) shows `uefi v0.35.0` as a direct dependency.

Any guidance on what might be causing this apparent failure to resolve standard `uefi-rs` prelude items and `#[entry]` macro behavior would be greatly appreciated.

nicholasbishop, 2025-06-24:
As was noted [here](https://github.com/rust-osdev/uefi-rs/issues/1708#issuecomment-2995386532), it would be much better to link to a small git repo that reproduces the issue. You've written a lot of text which makes it hard to see the actual problem. I wonder if perhaps you are using one of those genai tools? If so, I'd recommend you stop doing that, and read the [crate documentation](https://docs.rs/uefi/latest/uefi/) instead. I promise it'll give you vastly better info than an LLM :)

>  If efi_main takes no arguments, items from uefi::prelude::* like println!, image_handle(), system_table(), SystemTable, Boot should be resolved.

Indeed `efi_main` should take no arguments, this is documented in the [`entry` macro](https://docs.rs/uefi/latest/uefi/attr.entry.html).

See also the documentation for the [`prelude` module](https://docs.rs/uefi/latest/uefi/prelude/index.html). As you can see, `image_handle`, `system_table`, `SystemTable`, and `Boot` are not listed there. The latter two types were removed a while ago, see https://github.com/rust-osdev/uefi-rs/blob/main/docs/funcs_migration.md.

The `image_handle` function is here: https://docs.rs/uefi/latest/uefi/boot/fn.image_handle.html.

For the system table, if you need the raw pointer then use https://docs.rs/uefi/latest/uefi/table/fn.system_table_raw.html, or use the [`uefi::system`](https://docs.rs/uefi/latest/uefi/system/index.html) module for a more convenient wrapper.

General suggestions: read the documentation, read the compiler errors, and make sure you take the time to understand what documentation and compiler output are saying (an LLM can't do this work for you, you have to take the time to understand it yourself.)

deepm1nd, 2025-06-24:
Sorry, I used the LLM to write my issue because of my poor last showing. Won't do that again.