rust-osdev/uefi-rs #1733: uefi: update open protocol doc

phip1611, 2025-08-02:
> I have been trying to get non-exclusive access to a protocol as an app, but I don't understand the API yet
> 
> Use `boot::open_protocol` with [`OpenProtocolAttributes::GetProtocol`](https://docs.rs/uefi/latest/uefi/boot/enum.OpenProtocolAttributes.html#variant.GetProtocol). Here's an example: https://github.com/rust-osdev/uefi-rs/blob/d30f3c31036cea82fd8b13d6315affc9e420ee32/uefi-test-runner/src/proto/media.rs#L257-L264
> 
> 
> The documentation (and spec) are a little confusing since they say it's for drivers, but it works just as well with applications. (You can see in [`edk2`](https://github.com/tianocore/edk2/blob/7fe3609022ce7f52190bd6384d117f1510e15afe/MdeModulePkg/Core/Dxe/Hand/Handle.c#L1078) that `BY_HANDLE_PROTOCOL` and `GET_PROTOCOL` are handled the same way, and you can see example usages in popular UEFI applications such as [`shim`](https://github.com/rhboot/shim/blob/13ab598c2037b808f19e1ec795ec8f8f2789a54c/httpboot.c#L723) and [`grub`](https://github.com/rhboot/grub2/blob/6cac608cbe05b95ec2903897ad19dbd0499ab60d/grub-core/commands/efi/tpm.c#L167).) We should certainly update our docs to make this less confusing.
>  

 _Originally posted by @nicholasbishop in [#1725](https://github.com/rust-osdev/uefi-rs/issues/1725#issuecomment-3145107859)_

crawfxrd, 2025-11-26:
I had some trouble with this as well when porting code that used `HandleProtocol` (used everywhere in edk2), and originally ended up going through `uefi-raw` instead to use `handle_protocol` to keep it 1:1.

So IMO it would be nice to have:

- `uefi`
  - The code example using `GetProtocol` should be included in the docs
  - With a note that this is the replacement for `handle_protocol`
  - If possible, searching for `handle_protocol` should show `open_protocol` as a result
- `uefi-raw`
  - `handle_protocol` should add a note/warning to use `open_protocol` instead

crawfxrd, 2025-11-26:
Also potentially something for `LocateProtocol`, since that got removed as well. I used that in combination with `ReinstallProtocolInterface` to replace edk2 interfaces with my own. Now it's `get_handle_for_protocol()` + `open_protocol()`.

nicholasbishop, 2026-02-07:
I put up #1891 to improve the docs as discussed in the issue description. Also opened #1892 to cover the suggestion in https://github.com/rust-osdev/uefi-rs/issues/1733#issuecomment-3583237781.

closed 2026-02-08 by nicholasbishop