rust-osdev/uefi-rs #1754: Impossible to create an instance of or use Input struct in uefi::console::text::input

King-Caleb, 2025-09-03:
# Impossible to create an instance of or use Input struct in `uefi::console::text::input`

Hello,

I'm trying to read keyboard input using the latest `uefi` crate (`0.35.0`), but I ran into a problem with the `Input` struct in `uefi::console::text::input`.

The documentation for `Input` is very sparse, and it seems impossible to create an instance manually. For example, trying to do something like:

```rust
let input = Input::new(); // Doesn't exist
```

fails because the struct is opaque and has no public constructor. 

I also tried using the system table inside my entry function:

``` rust
#[entry]
fn kernel_main(handle: Handle, mut st: SystemTable<Boot>) -> Status {
    let input = st.stdin(); // returns &Input
    // ... but entry function can't have arguments
}
```

Even the docs examples don't work.

Could the maintainers clarify:

1. How to properly obtain and use an `Input` handle in `uefi 0.35.0`?  
2. Are there any working examples for reading keyboard input with the current version?  

Thank you for your time and for maintaining the crate!

nicholasbishop, 2025-09-03:
You can use [`uefi::system::with_stdin`](https://docs.rs/uefi/latest/uefi/system/fn.with_stdin.html) to use the `Input` protocol. I've uploaded a PR adding an example: https://github.com/rust-osdev/uefi-rs/pull/1755

For more general information on how to work with protocols, see:
* https://rust-osdev.github.io/uefi-rs/concepts/handles_and_protocols.html
* https://rust-osdev.github.io/uefi-rs/how_to/protocols.html

phip1611, 2025-09-07:
Closing as I currently don't see the bug here. If the example provided by Nicholas in #1755 doesn't help and we indeed have a bug, please comment here and we can open the issue again :)

closed 2025-09-07 by phip1611